schodet/nxt-python #177: Error struct.error: unpack requires a buffer of 2 bytes

Repich, 2022-01-19:
- OS: - Raspberry Pi OS
- Python version: 3.9.2
- NXT-Python version: 3.0.0a3
- Connection method: USB? Bluetooth? - Bluetooth

Hello. I'm using python 3.9, raspberry pi and EV3.
If disabled in Bluetooth setting support iPhone – code nxt_test --log-level=debug return
INFO:nxt.backend.bluetooth:connecting via Bluetooth (00:16:53:3D:89:40)
DEBUG:nxt.locator:found brick <nxt.brick.Brick object at 0xb64a6fe8>
DEBUG:nxt.backend.bluetooth:send: 0200019b
and freezes 

If support iPhone in Bluetooth setting enabled – code nxt_test --log-level=debug return
INFO:nxt.backend.bluetooth:connecting via Bluetooth (00:16:53:3D:89:40)
DEBUG:nxt.locator:found brick <nxt.brick.Brick object at 0xb6472fe8>
DEBUG:nxt.backend.bluetooth:send: 0200019b
DEBUG:nxt.backend.bluetooth:recv: ff
b'\xff'
INFO:nxt.backend.bluetooth:closing Bluetooth (00:16:53:3D:89:40) connection
Traceback (most recent call last):
  File "/home/pi/Documents/Python_Project/python-virtual-environments/env/bin/nxt_test", line 22, in <module>
    name, host, signal_strengths, user_flash = b.get_device_info()
  File "/home/pi/Documents/Python_Project/python-virtual-environments/env/lib/python3.9/site-packages/nxt/brick.py", line 1056, in get_device_info
    tgram = self._cmd(tgram)
  File "/home/pi/Documents/Python_Project/python-virtual-environments/env/lib/python3.9/site-packages/nxt/brick.py", line 328, in _cmd
    reply_tgram = Telegram(opcode=tgram.opcode, pkt=self._sock.recv())
  File "/home/pi/Documents/Python_Project/python-virtual-environments/env/lib/python3.9/site-packages/nxt/backend/bluetooth.py", line 81, in recv
    (plen,) = struct.unpack("<H", data)
struct.error: unpack requires a buffer of 2 bytes

I can't fix it

schodet, 2022-01-19:
Hello, thanks for the report.

What is “Bluetooth setting support iPhone”?

Can you run `bluetoothctl`, then issue the `devices` command and for each returned device, `info address_of_the_device`?

Something like this:

```
$ bluetoothctl
[bluetooth]# devices
Device 00:16:53:3D:89:40 NXT
Device B0:18:6F:EE:EE:EE JVC HA-S40BT
[bluetooth]# info 00:16:53:3D:89:40
Device 00:16:53:3D:89:40 (public)
        Name: NXT
        Alias: NXT
        Class: 0x00000804
        Paired: yes
        Trusted: no
        Blocked: no
        Connected: no
        LegacyPairing: no
        UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)
[bluetooth]# info B0:18:6F:EE:EE:EE
...
```

Thanks.

schodet, 2022-01-19:
I just realize: EV3

As far as I know, NXT-Python was never tested with EV3, only with NXT, do you have any reason to think this could work with EV3?

Repich, 2022-01-20:
> What is “Bluetooth setting support iPhone”?
It's setting on EV3
Photo device https://disk.yandex.ru/i/6xMAqv83KV4XEQ

> [bluetooth]# info 00:16:53:3D:89:40

[REPICH_EV3]# info 00:16:53:3D:89:40
Device 00:16:53:3D:89:40 (public)
	Name: REPICH_EV3
	Alias: REPICH_EV3
	Class: 0x00000804
	Paired: yes
	Trusted: no
	Blocked: no
	Connected: yes
	LegacyPairing: no
	UUID: Vendor specific           (00000000-deca-fade-deca-deafdecacaff)
	UUID: Serial Port               (00001101-0000-1000-8000-00805f9b34fb)

> As far as I know, NXT-Python was never tested with EV3, only with NXT, do you have any reason to think this could work with EV3?
Oh, understandable. I did not know, my knowledge is superficial, I thought it would work with my device.

schodet, 2022-01-20:
OK, closing then.

closed 2022-01-20 by schodet