schodet/nxt-python #180: Not able to download MotorControl22.nxc to NXT

Svein-Tore, 2022-02-28:
I have problems with downloading the Motorcontrol22.nxc to the NXT.  I get the following error message:

line 173:  Error: Unmatched preprocessor directive.

Line 173 is:

// if we exited from external NXTMotor.Stop command, we might've overwritten

Find it strange to give an error message since it is a comment......

Any suggestions?

Svein-Tore, 2022-03-01:
The problem is solved!  I downloaded the motorcontrol22.rxe from here:  

http://web5.uottawa.ca/www5/dcousineau/home/Others/Math4NXT/index.html 

I had to delete some of the files on the NXT to get the file downloaded.  But after that I got it work!! NICE work Nicholas!   

PS I recommend to add the .rcx file here on gihtub.

closed 2022-03-01 by Svein-Tore

schodet, 2022-03-01:
Glad you found a way to make it work. I open an issue [here](https://github.com/schodet/MotorControl/issues/1) to add the pre-compiled file when I get back to my computer.