schodet/nxt-python #181: nxt_filer needs to be updated (minor fixes)

tcwan, 2022-03-16:
nxt_filer script is no longer tracking the latest nxt-python 3.0.0 and Gtk library requirements. This is a minor patch to fix the bugs.

[nxt_filer.diff.txt](https://github.com/schodet/nxt-python/files/8260053/nxt_filer.diff.txt)

schodet, 2022-03-16:
How did I missed this? Thanks for the patch.

schodet, 2022-03-16:
The import and flags arguments can actually be removed.

Are you OK with this commit:

```diff
commit 79d710b1f2391644567fc3223b372c817893693c (HEAD -> master)
Author: TC Wan <tcwan@usm.my>
Date:   Wed Mar 16 09:20:35 2022 +0100

    Fix nxt_filer

diff --git a/scripts/nxt_filer b/scripts/nxt_filer
index c4cbdb6..6839022 100755
--- a/scripts/nxt_filer
+++ b/scripts/nxt_filer
@@ -57,7 +57,6 @@ import nxt.locator
 from usb.core import USBError
 from nxt.locator import BrickNotFoundError
 from nxt.error import FileExistsError, SystemProtocolError
-from nxt.telegram import OPCODES, Telegram
 
 
 def play_soundfile(b, fname):
@@ -382,9 +381,7 @@ class AboutDialog(Gtk.AboutDialog):
 
 class Application(Gtk.Application):
     def __init__(self, *args, **kwargs):
-        super().__init__(*args,
-                         flags=Gio.ApplicationFlags.HANDLES_COMMAND_LINE,
-                         **kwargs)
+        super().__init__(*args, **kwargs)
         self.window = None
 
         self.set_option_context_summary(
```

tcwan, 2022-03-17:
I'm fine with it.

closed 2022-03-17 by schodet