schodet/nxt-python #182: usb.core.USBError: [Errno None] b'libusb0-dll:err [set_configuration] could not set config 1

MrCheatEugene, 2022-04-19:
- OS: Windows 10
- Python version: 3.9 64 bit
- NXT-Python version: 3.0.1
- Connection method: USB? Bluetooth? USB

Hi there. I am trying to connect my NXT brick to python using this library. But, it wasn't detecting it. I have updated my firmware to 1.31, but that didn't help. Also, I have found an article that said that I need to install libusb filters. But after installing them, the test program doesn't want to work at all.
Here is the error:
```Traceback (most recent call last):
  File "C:\Users\username\test\test.py", line 2, in <module>
    with nxt.locator.find(name="NXT") as b:
  File "C:\Users\username\AppData\Local\Programs\Python\Python39\lib\site-packages\nxt\locator.py", line 213, in find
    brick = next(iter_bricks(), None)
  File "C:\Users\username\AppData\Local\Programs\Python\Python39\lib\site-packages\nxt\locator.py", line 191, in iter_bricks
    for brick in backend.find(name=name, host=host, **filters):
  File "C:\Users\username\AppData\Local\Programs\Python\Python39\lib\site-packages\nxt\backend\usb.py", line 102, in find
    brick = sock.connect()
  File "C:\Users\username\AppData\Local\Programs\Python\Python39\lib\site-packages\nxt\backend\usb.py", line 57, in connect
    self._dev.set_configuration()
  File "C:\Users\username\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\core.py", line 915, in set_configuration
    self._ctx.managed_set_configuration(self, configuration)
  File "C:\Users\username\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\username\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\core.py", line 159, in managed_set_configuration
    self.backend.set_configuration(self.handle, cfg.bConfigurationValue)
  File "C:\Users\username\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\backend\libusb0.py", line 509, in set_configuration
    _check(_lib.usb_set_configuration(dev_handle, config_value))
  File "C:\Users\username\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\backend\libusb0.py", line 447, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [set_configuration] could not set config 1: win error: \xcf\xe0\xf0\xe0\xec\xe5\xf2\xf0 \xe7\xe0\xe4\xe0\xed \xed\xe5\xe2\xe5\xf0\xed\xee.\r\n'```
The symbols that start with \x are the symbols of my language, because windows reports errors on user's language.
Sorry for that.

Why is this happening? How can I fix this?

schodet, 2022-04-19:
What are those libusb filters? What is the article you refer to?

Can you translate the error message given  in your langage?

Thanks.

MrCheatEugene, 2022-04-19:
Hi. I don't know, the article said that I need to install it, so I did.
The article is on russian, use translator please:
http://nnxt.blogspot.com/2012/11/nxt-python-windows.html
I would do that, If I could actually read it.. Can you give me decoded
text? I don't know what encoding it is.
Greetings, Emil.

MrCheatEugene, 2022-04-19:
Also in the article they used Python 2, but I used python 3. Also I tried doing the beep with old version of this library that I have found somewhere in the internet, and it worked perfectly. I've needed this library for a project, but it would be nice if I can somehow bypass this error and work with fresh library and not release a project with 2 different versions of python(first part is on Python 3.9 and the nxt part is on 2.7)

schodet, 2022-04-20:
I tried this:

```
>>> b"\xcf\xe0\xf0\xe0\xec\xe5\xf2\xf0 \xe7\xe0\xe4\xe0\xed \xed\xe5\xe2\xe5\xf0\xed\xee".decode("cp1251")
'Параметр задан неверно'
>>>
```

According to translator: “Parameter set incorrectly”.

Can you try to comment out `self._dev.set_configuration()` in nxt/backend/usb.py and see if it works?

MrCheatEugene, 2022-04-21:
> I tried this:
>
> ```
> 'Параметр задан неверно'
> ```
>
> According to translator: “Parameter set incorrectly”.
>
> Can you try to comment out `self._dev.set_configuration()` in nxt/backend/usb.py and see if it [...]

Hi. Now it outputs this:
``` Traceback (most recent call last):
  File "C:\Windows.old\Users\Username\Desktop\test\test.py", line 4, in <module>
    with nxt.locator.find() as b:
  File "C:\Users\Username\AppData\Local\Programs\Python\Python39\lib\site-packages\nxt\locator.py", line 213, in find
    brick = next(iter_bricks(), None)
  File "C:\Users\Username\AppData\Local\Programs\Python\Python39\lib\site-packages\nxt\locator.py", line 191, in iter_bricks
    for brick in backend.find(name=name, host=host, **filters):
  File "C:\Users\Username\AppData\Local\Programs\Python\Python39\lib\site-packages\nxt\backend\usb.py", line 102, in find
    brick = sock.connect()
  File "C:\Users\Username\AppData\Local\Programs\Python\Python39\lib\site-packages\nxt\backend\usb.py", line 58, in connect
    intf = self._dev.get_active_configuration()[(0, 0)]
  File "C:\Users\Username\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\core.py", line 921, in get_active_configuration
    return self._ctx.get_active_configuration(self)
  File "C:\Users\Username\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\core.py", line 113, in wrapper
    return f(self, *args, **kwargs)
  File "C:\Users\Username\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\core.py", line 252, in get_active_configuration
    bConfigurationValue=self.backend.get_configuration(self.handle)
  File "C:\Users\Username\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\backend\libusb0.py", line 519, in get_configuration
    ret = self.ctrl_transfer(
  File "C:\Users\Username\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\backend\libusb0.py", line 601, in ctrl_transfer
    return _check(_lib.usb_control_msg(
  File "C:\Users\Username\AppData\Local\Programs\Python\Python39\lib\site-packages\usb\backend\libusb0.py", line 447, in _check
    raise USBError(errmsg, ret)
usb.core.USBError: [Errno None] b'libusb0-dll:err [control_msg] sending control message failed, win error: \xcd\xe5 \xf3\xe4\xe0\xe5\xf2\xf1\xff \xed\xe0\xe9\xf2\xe8 \xf3\xea\xe0\xe7\xe0\xed\xed\xfb\xe9 \xf4\xe0\xe9\xeb.\r\n\n'
```
And I still have no idea how to convert these symbols into text that I can translate.

I've commented the reset line as you said, but left configuration sit still and it worked! Thanks!

schodet, 2022-04-21:
To convert, i pasted the error on a python3 interpreter and used the decode method:

```
% python3
Python 3.9.2 (default, Feb 28 2021, 17:03:44) 
[GCC 10.2.1 20210110] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> b'\xcd\xe5 \xf3\xe4\xe0\xe5\xf2\xf1\xff \xed\xe0\xe9\xf2\xe8 \xf3\xea\xe0\xe7\xe0\xed\xed\xfb\xe9 \xf4\xe0\xe9\xeb'.decode('cp1251')
'Не удается найти указанный файл'
>>>
```

I can add a workaround for windows, I have seen old code skipping the reset for Windows.

@dbrown3562 could you confirm that you also need to comment out `reset` to make it work on Windows?

closed 2022-04-22 by schodet