schodet/nxt-python #191: MotCont: Differential drive robot

tsampazk, 2022-12-01:
It is stated that synchronized driving is possible with motcont. 

My understanding is that one can send a single command to control both motors so they turn simultaneously by providing a list of ports in `cmd`, but is there a way to send a command with something like a "ratio" so a differential drive robot can turn? I found some comments regarding this in #47.

If there isn't a way to send such a command, is there another standard way to implement this?

schodet, 2022-12-02:
I do not think that MotorControl support the turn ratio. This would be an interesting feature to add.

You can use [nxt.motor.SynchronizedMotors](https://ni.srht.site/nxt-python/latest/api/motor.html#nxt.motor.SynchronizedMotors), but the regulation is not that good. Give it a try.

tsampazk, 2022-12-02:
Thank you! 

To be honest, I would prefer to have precise control of the motors, that's why i went with motcont. I am trying to find a way to implement a turn ratio, i will share the results if it works well. If it doesn't, i will probably revert to doing simple turns by using only one motor each time or use nxt.motor.SynchronizedMotors.

If it isn't too much trouble, could you provide a general high level explanation of how turn ratio is implemented in nxt.motor.SynchronizedMotors? By examining the code i can only get to the point where the turn ratio is passed in the state? Is it handled internally in the nxt somehow?

schodet, 2022-12-06:
This is done by the firmware, you can find the related code in https://git.ni.fr.eu.org/nxt-firmware.git/tree/AT91SAM7S256/Source/d_output.c?h=1.29.2

For a differential drive robot, what I find is the best method is to control the sum and the difference of the tacho of the two motors. The sum controls linear movement while the difference controls angular movement. This allows to drive really straight.

If you change MotorControl to implement this, that would be great!

I suppose you have this yet: https://github.com/schodet/MotorControl

tsampazk, 2022-12-06:
> I suppose you have this yet: https://github.com/schodet/MotorControl

Yes i have been using MotorControl for some days now! It is indeed really useful for precise motor control.

Thank you very much for the information, i will be working on this in the following weeks. 

> If you change MotorControl to implement this, that would be great!

I will try to make something generic that might be useful for motor control and if so i will contribute to the repository!

closed 2022-12-06 by tsampazk