schodet/nxt-python #193: Bluetooth Error

Josh-Hout, 2023-03-13:
- OS: Raspberry pi OS
- Python version: Python 3.9.2
- NXT-Python version: I believe python-3.2.0
- Connection method: Bluetooth

Im Trying to run code over Bluetooth

this is my simple program found in the nxt doc

```
import nxt.locator
from nxt.motor import *

#00:16:53:15:DC:F2 is my NXTs bluetooth MAC adress

with nxt.locator.find(host="00:16:53:15:DC:F2") as b:
    print(b.get_device_info()[0:2])
```

I ran the program and got a result of
```
failed to connect to device Bluetooth (00:16:53:15:DC:F2)
Traceback (most recent call last):
  File "/home/joshua/Mindstorms/Log.py", line 6, in <module>
    with nxt.locator.find(host="00:16:53:15:DC:F2") as b:
  File "/home/joshua/.local/lib/python3.9/site-packages/nxt/locator.py", line 215, in find
    raise BrickNotFoundError("no brick found")
nxt.locator.BrickNotFoundError: no brick found
```

And i Do have python3-bluez

Am i doing something wrong?

Josh-Hout, 2023-03-13:
Running with Debug i got more info
```
DEBUG:nxt.locator:configuration files=['.nxt-python.conf', '/home/joshua/.nxt-python.conf']
DEBUG:nxt.locator:configuration read from []
DEBUG:nxt.locator:no section default, using DEFAULT
INFO:nxt.locator:using backend from nxt.backend.usb
INFO:nxt.locator:using backend from nxt.backend.bluetooth
INFO:nxt.backend.bluetooth:connecting via Bluetooth (00:16:53:15:DC:F2)
WARNING:nxt.backend.bluetooth:failed to connect to device Bluetooth (00:16:53:15:DC:F2)
DEBUG:nxt.backend.bluetooth:error from connect
Traceback (most recent call last):
  File "<string>", line 3, in connect
_bluetooth.error: (111, 'Connection refused')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/joshua/.local/lib/python3.9/site-packages/nxt/backend/bluetooth.py", line 127, in find
    brick = sock.connect()
  File "/home/joshua/.local/lib/python3.9/site-packages/nxt/backend/bluetooth.py", line 52, in connect
    sock.connect((self._host, PORT))
  File "<string>", line 5, in connect
bluetooth.btcommon.BluetoothError: [Errno 111] Connection refused
```

Zackariyya, 2023-03-15:
Did you pair the NXT with the Raspberry PI ?

schodet, 2023-03-16:
Is your NXT paired with your computer?

How did you get the device address?

I should add something in the manual for a step by step procedure to connect for each platform.

Josh-Hout, 2023-03-16:
I got the device address by using usb but then I realized that won’t work because my robot would move

schodet, 2023-03-20:
Is your NXT paired with your computer?

Josh-Hout, 2023-03-20:
I can only connect with serial port

Josh-Hout, 2023-03-20:
But it is connected

dmue-vbg, 2023-04-14:
**got similar issue (on macos) :**

DEBUG:nxt.locator:configuration read from []
DEBUG:nxt.locator:no section default, using DEFAULT
INFO:nxt.locator:using backend from nxt.backend.usb

**INFO:nxt.backend.bluetooth:no bluetooth module**

Traceback (most recent call last):
  File "/Users/dennis/Desktop/nxt_test.py", line 16, in <module>
    with nxt.locator.find(host="00:16:53:1B:AC:97") as b:
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/nxt/locator.py", line 215, in find
    raise BrickNotFoundError("no brick found")
nxt.locator.BrickNotFoundError: no brick found

schodet, 2023-04-21:
@dmue-vbg 
> **INFO:nxt.backend.bluetooth:no bluetooth module**

That’s not the same problem, you do not have a python bluetooth module installed.

schodet, 2024-10-06:
Closing, no news from reporter.

closed 2024-10-06 by schodet