schodet/nxt-python #195: simulate NXT brick button presses with nxt-python

arohrbacher, 2023-12-17:
Hi,

I am wondering if it is possible to "simulate" NXT brick button presses with nxt-python, i.e. issue a command in Python that acts as if a physical button the the brick has been pressed or causes the NXT brick to register a button press. According to the nxt-python documentation there are several low-level functions available, but I didn't find anything pertaining to the buttons.
I am using nxt-screen.py (because my NXT brick has a non-working screen :-) ), so at least the physical screen (or framebuffer) of a NXT brick can be read; it would be cool if the brick could be remotely controlled with something like nxt-screen.py, but with added button control...

Thanks,

Andreas

schodet, 2023-12-19:
You can access internal firmware using io maps, there is an io map for button here:

https://git.ni.fr.eu.org/nxt-firmware.git/tree/src/c_button.iom

You can try to make sense of the corresponding module to see whether it would be possible to simulate a button press:

https://git.ni.fr.eu.org/nxt-firmware.git/tree/src/c_button.c

Here is the function used to read the button state for the menus:

https://git.ni.fr.eu.org/nxt-firmware.git/tree/src/c_ui.c#n389

So setting PRESSED_STATE in the corresponding State array should work.

schodet, 2023-12-19:
Tell me if you need help to make a prototype.

arohrbacher, 2023-12-20:
Hi Nicolas,

many thanks for your pointers - very helpful, I got it to work!
First I had to figure out the module id for the buttons: this is did with
`modules = b.find_modules()for x in modules:
print(x)`

arohrbacher, 2023-12-20:
Hi Nicolas,
many thanks for your pointers! Very helpful, I got it to work!

First I had to figure out the module id for the buttons:
`    modules = b.find_modules()`
`    for x in modules:`
`        print(x)`
resulting in (amongs others): `('Button.mod', 262145, 0, 36)`
I was then continuously reading the iomap data with
`    data = b.read_io_map(0x40001, 0, 36)[1]`
`    for y in range(0,len(data)):`
`        print(data[y], end = ' ')`
`    print()`
and observing what happens when I press the buttons on the NXT brick. After I knew this, I could issue this group of commands
`    b.write_io_map(0x40001, 34, bytearray([0x80]))`
`    b.write_io_map(0x40001, 34, bytearray([0x80]))`
`    b.write_io_map(0x40001, 34, bytearray([0x0]))`
`    b.write_io_map(0x40001, 34, bytearray([0x0]))`
which "simulates" a press of the left-button (complete with "!Click"-sound). The right-button would have offset 33, return-button offset 32, and exit-button offset 35.
For some reason I have to issue the write_io_map commands twice - otherwise my Python program would "miss" a simulated keystroke...
Anyway, now I just have to extend nxt-screen.py to include a button-interface to get a completely remote-controlled NXT brick.

Best regards,

Andreas

schodet, 2023-12-20:
Nice, waiting for your pull request :)

closed 2023-12-20 by schodet