schodet/nxt-python #199: PyBluez is not under development

Zackariyya, 2024-12-20:
Consider some alternatives:

[Bleak](https://github.com/hbldh/bleak): Cross-platform Bluetooth Low Energy client library
[Bless](https://github.com/kevincar/bless): Cross-platform Bluetooth Low Energy server library
[PyQt](https://www.riverbankcomputing.com/software/pyqt/): Python bindings for cross-platform [QtBluetooth](https://www.riverbankcomputing.com/static/Docs/PyQt6/api/qtbluetooth/qtbluetooth-module.html) API

schodet, 2024-12-20:
The NXT is using Bluetooth classic, not BLE.

QT might be an option, would you like to try it?

dlech, 2024-12-20:
Since the NXT just uses RFCOMM, the builtin Bluetooth sockets in Python should work for Windows and Linux. Then for Mac, we can use PyObjC which has IOBluetooth bindings that include RFCOMM support.

schodet, 2024-12-20:
> Since the NXT just uses RFCOMM, the builtin Bluetooth sockets in Python should work for Windows and Linux. Then for Mac, we can use PyObjC which has IOBluetooth bindings that include RFCOMM support.

That would be a good option. The discovery might be missing, if I remember correctly we are supposed to use dbus under Linux.

The PyObjC option remind me of something… maybe while reading pybluez source code.

schodet, 2024-12-21:
I made a first proof of concept using python socket on the btsocket branch here or at srht.

It works on my Debian bookworm, if you can give it a try on Fedora 41, Windows and MacOS, this would be nice, documentation is not very verbose on platform support.

```
nicolas@boktor:~/work/lego/nxt-python% poetry run python scripts/nxt_test --backend btsocket  --log-level=DEBUG --host 00:16:53:AA:BB:CC
Find brick...
DEBUG:nxt.locator:configuration files=['.nxt-python.conf', '/home/nicolas/.nxt-python.conf']
DEBUG:nxt.locator:configuration read from []
DEBUG:nxt.locator:no section default, using DEFAULT
INFO:nxt.locator:using backend from nxt.backend.btsocket
INFO:nxt.backend.btsocket:connecting via BtSocket (00:16:53:AA:BB:CC)
DEBUG:nxt.locator:found brick <nxt.brick.Brick object at 0x7fec9d6cf890>
DEBUG:nxt.backend.btsocket:send: 0200019b
DEBUG:nxt.backend.btsocket:recv: 2100
DEBUG:nxt.backend.btsocket:recv: 029b004e5854320000000000000000000000001653AABBCC000000000074320100
DEBUG:nxt.locator:found brick with name=NXT2 and host=00:16:53:08:91:33
DEBUG:nxt.backend.btsocket:send: 0200019b
DEBUG:nxt.backend.btsocket:recv: 2100
DEBUG:nxt.backend.btsocket:recv: 029b004e5854320000000000000000000000001653AABBCC000000000074320100
NXT brick name: NXT2
Host address: 00:16:53:AA:BB:CC
Bluetooth signal strengths: (0, 0, 0, 0)
Free user flash: 78452
DEBUG:nxt.backend.btsocket:send: 02000188
DEBUG:nxt.backend.btsocket:recv: 0700
DEBUG:nxt.backend.btsocket:recv: 0288007c011d01
Protocol version 1.124
Firmware version 1.29
DEBUG:nxt.backend.btsocket:send: 0200000b
DEBUG:nxt.backend.btsocket:recv: 0500
DEBUG:nxt.backend.btsocket:recv: 020b001a21
Battery level 8474 mV
Play test sound...DEBUG:nxt.backend.btsocket:send: 060080032c013200
DEBUG:nxt.backend.btsocket:send: 0600800390013200
DEBUG:nxt.backend.btsocket:send: 06008003f4013200
DEBUG:nxt.backend.btsocket:send: 0600800358023200
done
INFO:nxt.backend.btsocket:closing BtSocket (00:16:53:AA:BB:CC) connection
nicolas@boktor:~/work/lego/nxt-python%
```

Thanks.