schodet/nxt-python #200: bluetooth.btcommon.BluetoothError: [Errno 14] Bad address

Zackariyya, 2024-12-20:
- OS: Fedora 41
- Python version: Python 3.13.0
- NXT-Python version:3.3.0
- Connection method: Bluetooth

```
Traceback (most recent call last):
  File "<string>", line 3, in send
_bluetooth.error: (14, 'Bad address')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/mnt/DATA/CODING_FILES/MicroControllers/NXT/nxt-car-tkinter__V2.py", line 30, in <module>
    print("Found brick:", b.get_device_info()[0])
                          ~~~~~~~~~~~~~~~~~^^
  File "/home/$USERNAME/.local/lib/python3.13/site-packages/nxt/brick.py", line 1056, in get_device_info
    tgram = self._cmd(tgram)
  File "/home/$USERNAME/.local/lib/python3.13/site-packages/nxt/brick.py", line 326, in _cmd
    self._sock.send(tgram.bytes())
    ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/home/$USERNAME/.local/lib/python3.13/site-packages/nxt/backend/bluetooth.py", line 70, in send
    self._sock.send(data)
    ~~~~~~~~~~~~~~~^^^^^^
  File "<string>", line 5, in send
bluetooth.btcommon.BluetoothError: [Errno 14] Bad address

```

schodet, 2024-12-20:
Do you have more details?

What’s your program? What parameters did you use? Does `bluetoothctl` manage to see the brick?

Zackariyya, 2024-12-20:
This is my program
```
#!/usr/bin/python3
import nxt.locator
import tkinter as tk

def onKeyPress(event):
    e = event.char
    if e == 'w':
        print("Forward")
        wheel1.run(-100)
        wheel2.run(-100)

    if e == 'd':
        print("Left")
        steering.turn(25,60)
    if e == 's':
        print("Back")
        wheel1.run(100)
        wheel2.run(100)
    if e == 'a':
        print("Right")
        steering.turn(-25,60)
    if e == 'q':
        print("Stop Motor")
        wheel1.brake()
        wheel2.brake()

def onKeyRelease(event):
    print("Released "+event)
    e = event.char
    if e == 'w':
        print("STOP Forward")
        wheel1.brake()
        wheel1.brake()

    if e == 'd':
        print("UNDO Left")
        steering.turn(-25,60)
    if e == 's':
        print("STOP Back")
        wheel1.brake()
        wheel1.brake()
    if e == 'a':
        print("UNDO Right")
        steering.turn(25,60)
    if e == 'q':
        print("Release Motor")
        wheel1.coast()
        wheel2.coast()
# Find a brick.
with nxt.locator.find() as b:
    # Once found, print its name.
    print("Found brick:", b.get_device_info()[0])
    # And play a recognizable note.
    b.play_tone(440, 250)
    
    # Get the motor connected to the port A.
    steering = b.get_motor(nxt.motor.Port.A)
    wheel1 = b.get_motor(nxt.motor.Port.B)
    wheel2 = b.get_motor(nxt.motor.Port.C)
    # Full circle in one direction.
    steering.turn(25, 60)
    steering.turn(-25, 60)  
    
    root = tk.Tk()
    root.geometry('300x200')
    text = tk.Label(root, text="Focus this window and use WASD to control NXT")
    text.pack()
    root.bind('<KeyPress>', onKeyPress)
    root.bind('<KeyRelease>', onKeyRelease)
    root.mainloop()
    steering.idle()
    wheel.idle()
```
bluetoothctl connects to brick, then after a few seconds it disconnects : 
```
USERNAME:~/ $ bluetoothctl                                                                                                                                                                                   [11:38:31]
[bluetooth]# Agent registered
[NXTI]# [CHG] Device 00:12:34:56:78:AB Connected: yes
[NXT]# [CHG] Device 00:12:34:56:78:AB ServicesResolved: yes
[NXT]# [CHG] Device 00:12:34:56:78:AB ServicesResolved: no
[bluetooth]# [CHG] Device 00:12:34:56:78:AB Connected: no
[bluetooth]# 
```
output of `` nxt_test --log-level=debug`` : 
```
Find brick...
DEBUG:nxt.locator:configuration files=['.nxt-python.conf', '/home/USERNAME/.nxt-python.conf']
DEBUG:nxt.locator:configuration read from []
DEBUG:nxt.locator:no section default, using DEFAULT
INFO:nxt.locator:using backend from nxt.backend.usb
INFO:nxt.locator:using backend from nxt.backend.bluetooth
INFO:nxt.backend.bluetooth:connecting via Bluetooth (00:12:34:56:78:AB)
DEBUG:nxt.locator:found brick <nxt.brick.Brick object at 0x7ff067aef8c0>
DEBUG:nxt.backend.bluetooth:send: 0200019b
INFO:nxt.backend.bluetooth:closing Bluetooth (00:12:34:56:78:AB) connection
Traceback (most recent call last):
  File "<string>", line 3, in send
_bluetooth.error: (14, 'Bad address')

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/USERNAME/.local/bin/nxt_test", line 26, in <module>
    name, host, signal_strengths, user_flash = b.get_device_info()
                                               ~~~~~~~~~~~~~~~~~^^
  File "/home/USERNAME/.local/lib/python3.13/site-packages/nxt/brick.py", line 1056, in get_device_info
    tgram = self._cmd(tgram)
  File "/home/zackariyyasattaur/.local/lib/python3.13/site-packages/nxt/brick.py", line 326, in _cmd
    self._sock.send(tgram.bytes())
    ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/home/USERNAME/.local/lib/python3.13/site-packages/nxt/backend/bluetooth.py", line 70, in send
    self._sock.send(data)
    ~~~~~~~~~~~~~~~^^^^^^
  File "<string>", line 5, in send
bluetooth.btcommon.BluetoothError: [Errno 14] Bad address
```
While that is running, bluetoothctl reports that it connects, then immediatly disconnects:
```
[NXT]# [CHG] Device 00:12:34:56:78:AB Connected: yes
[bluetooth]# [CHG] Device 00:12:34:56:78:AB Connected: no
```
NOTE: Those are not the real MAC addresses, I replaced them with that random one.

schodet, 2024-12-20:
Does your address starts with 00:16:53?

Make sure there is no NXT-Python program still running in the background (reboot if not sure).

Here is what I do to make it work:

```
% bluetoothctl
Agent registered
[bluetooth]# devices
[bluetooth]# scan on
Discovery started
[NEW] Device 00:16:53:AA:BB:CC 00-16-53-AA-BB-CC
[CHG] Device 00:16:53:AA:BB:CC LegacyPairing: no
[CHG] Device 00:16:53:AA:BB:CC Name: NXT1
[CHG] Device 00:16:53:AA:BB:CC Alias: NXT1
[CHG] Device 00:16:53:AA:BB:CC LegacyPairing: yes
[bluetooth]# pair 00:16:53:AA:BB:CC
Attempting to pair with 00:16:53:AA:BB:CC
Request PIN code
(here, enter code 1234 on NXT screen)
[agent] Enter PIN code: 1234
[CHG] Device 00:16:53:AA:BB:CC Connected: yes
[CHG] Device 00:16:53:AA:BB:CC UUIDs: 00001101-0000-1000-8000-00805f9b34fb
[CHG] Device 00:16:53:AA:BB:CC ServicesResolved: yes
[CHG] Device 00:16:53:AA:BB:CC Paired: yes
Pairing successful
[CHG] Device 00:16:53:AA:BB:CC ServicesResolved: no
[CHG] Device 00:16:53:AA:BB:CC Connected: no
[bluetooth]# scan off
Discovery stopped
[bluetooth]# devices
Device 00:16:53:AA:BB:CC NXT1
[bluetooth]#
```

Now I can run a nxt-python script using the 00:16:53:AA:BB:CC address.

If I keep bluetoothctl open, during the script run, I can see:

```
[CHG] Device 00:16:53:AA:BB:CC Connected: yes
[CHG] Device 00:16:53:AA:BB:CC Connected: no
```

So this part is normal.

If it does not work, try to remove the NXT from known devices and start again.

Zackariyya, 2024-12-21:
I did the steps and got the same exact error
My address does start with 16:53

schodet, 2024-12-21:
There is an open ticket about this: https://github.com/pybluez/pybluez/issues/473

Could you do this test, replacing the address with your NXT address:

```
strace nxt_test --host 00:16:53:AA:BB:CC
```

Thanks.

schodet, 2024-12-21:
There is another option, using the devfile backend:

```
nicolas@boktor:~/work/lego/nxt-python% sudo rfcomm bind 0 00:16:53:AA:BB:CC
[sudo] password for nicolas: 
nicolas@boktor:~/work/lego/nxt-python% ls /dev/rfcomm0 
/dev/rfcomm0
nicolas@boktor:~/work/lego/nxt-python% nxt_test --backend devfile
Find brick...
NXT brick name: NXT2
Host address: 00:16:53:AA:BB:CC
Bluetooth signal strengths: (0, 0, 0, 0)
Free user flash: 78452
Protocol version 1.124
Firmware version 1.29
Battery level 8433 mV
Play test sound...done
nicolas@boktor:~/work/lego/nxt-python% nxt_test --backend devfile --log-level DEBUG
Find brick...
DEBUG:nxt.locator:configuration files=['.nxt-python.conf', '/home/nicolas/.nxt-python.conf']
DEBUG:nxt.locator:configuration read from []
DEBUG:nxt.locator:no section default, using DEFAULT
INFO:nxt.locator:using backend from nxt.backend.devfile
INFO:nxt.backend.devfile:connecting via /dev/rfcomm0
DEBUG:nxt.locator:found brick <nxt.brick.Brick object at 0x7fe734361d10>
DEBUG:nxt.backend.devfile:send: 0200019b
DEBUG:nxt.backend.devfile:recv: 2100
DEBUG:nxt.backend.devfile:recv: 029b004e5854320000000000000000000000001653AABBCC000000000074320100
NXT brick name: NXT2
Host address: 00:16:53:AA:BB:CC
Bluetooth signal strengths: (0, 0, 0, 0)
Free user flash: 78452
DEBUG:nxt.backend.devfile:send: 02000188
DEBUG:nxt.backend.devfile:recv: 0700
DEBUG:nxt.backend.devfile:recv: 0288007c011d01
Protocol version 1.124
Firmware version 1.29
DEBUG:nxt.backend.devfile:send: 0200000b
DEBUG:nxt.backend.devfile:recv: 0500
DEBUG:nxt.backend.devfile:recv: 020b00f120
Battery level 8433 mV
Play test sound...DEBUG:nxt.backend.devfile:send: 060080032c013200
DEBUG:nxt.backend.devfile:send: 0600800390013200
DEBUG:nxt.backend.devfile:send: 06008003f4013200
DEBUG:nxt.backend.devfile:send: 0600800358023200
done
INFO:nxt.backend.devfile:closing /dev/rfcomm0 connection
nicolas@boktor:~/work/lego/nxt-python% sudo rfcomm release 0
nicolas@boktor:~/work/lego/nxt-python%
```

Zackariyya, 2024-12-22:
``strace nxt_test --host 00:16:53:AA:BB:C``
https://pastebin.com/e5xZwwTi

Zackariyya, 2024-12-22:
rfcomm doesn't work, it just hangs until i kill it : 
```
Find brick...
DEBUG:nxt.locator:configuration files=['.nxt-python.conf', '/home/zackariyyasattaur/.nxt-python.conf']
DEBUG:nxt.locator:configuration read from ['.nxt-python.conf', '/home/zackariyyasattaur/.nxt-python.conf']
INFO:nxt.locator:using backend from nxt.backend.devfile
INFO:nxt.backend.devfile:connecting via /dev/rfcomm0
```
tio just connects and disconnects:
```
[12:33:12.259] tio 3.8
[12:33:12.259] Press ctrl-t q to quit
[12:33:12.259] Connected to /dev/rfcomm0
[12:33:13.139] Disconnected
[12:33:13.473] Connected to /dev/rfcomm0
[12:33:15.600] Disconnected
[12:33:16.602] Connected to /dev/rfcomm0
[12:33:18.164] Disconnected
[12:33:19.165] Connected to /dev/rfcomm0
[12:33:19.438] Disconnected
[12:33:20.440] Connected to /dev/rfcomm0
[12:33:20.726] Disconnected
[12:33:21.727] Connected to /dev/rfcomm0
[12:33:22.100] Disconnected
[12:33:23.105] Connected to /dev/rfcomm0
[12:33:24.563] Disconnected
[12:33:25.564] Connected to /dev/rfcomm0
[12:33:25.838] Disconnected
[12:33:26.840] Connected to /dev/rfcomm0
[12:33:27.125] Disconnected
[12:33:28.127] Connected to /dev/rfcomm0
[12:33:28.400] Disconnected
[12:33:29.402] Connected to /dev/rfcomm0
[12:33:29.676] Disconnected
[12:33:30.676] Connected to /dev/rfcomm0
[12:33:30.963] Disconnected
[12:33:31.965] Connected to /dev/rfcomm0
[12:33:32.238] Disconnected
[12:33:33.239] Connected to /dev/rfcomm0
[12:33:33.526] Disconnected
[12:33:34.527] Connected to /dev/rfcomm0
[12:33:34.814] Disconnected
[12:33:35.816] Connected to /dev/rfcomm0
[12:33:37.363] Disconnected
[12:33:38.364] Connected to /dev/rfcomm0
[12:33:38.750] Disconnected
[12:33:39.752] Connected to /dev/rfcomm0
[12:33:41.201] Disconnected
[12:33:42.202] Connected to /dev/rfcomm0
[12:33:42.476] Disconnected
[12:33:43.477] Connected to /dev/rfcomm0
[12:33:43.763] Disconnected
[12:33:44.765] Connected to /dev/rfcomm0
[12:33:45.039] Disconnected
[12:33:46.040] Connected to /dev/rfcomm0
[12:33:46.326] Disconnected
[12:33:47.327] Connected to /dev/rfcomm0
[12:33:47.601] Disconnected
[12:33:48.602] Connected to /dev/rfcomm0
[12:33:48.889] Disconnected
[12:33:49.890] Connected to /dev/rfcomm0
[12:33:50.164] Disconnected
[12:33:51.165] Connected to /dev/rfcomm0
[12:33:51.451] Disconnected
[12:33:52.452] Connected to /dev/rfcomm0
[12:33:52.726] Disconnected
[12:33:53.727] Connected to /dev/rfcomm0
[12:33:54.001] Disconnected
[12:33:55.003] Connected to /dev/rfcomm0
[12:33:55.288] Disconnected
[... 145 lines cut ...]
[12:35:38.978] Disconnected
[12:35:39.980] Connected to /dev/rfcomm0
[12:35:40.253] Disconnected
[12:35:41.254] Connected to /dev/rfcomm0
[12:35:41.515] Disconnected
[12:35:42.516] Connected to /dev/rfcomm0
[12:35:42.804] Disconnected
[12:35:43.806] Connected to /dev/rfcomm0
[12:35:44.078] Disconnected
[12:35:45.079] Connected to /dev/rfcomm0
[12:35:45.366] Disconnected
[12:35:46.366] Connected to /dev/rfcomm0
[12:35:46.641] Disconnected
[12:35:47.643] Connected to /dev/rfcomm0
[12:35:47.929] Disconnected
[12:35:48.930] Connected to /dev/rfcomm0
[12:35:49.204] Disconnected
[12:35:50.206] Connected to /dev/rfcomm0
[12:35:50.479] Disconnected
[12:35:51.481] Connected to /dev/rfcomm0
[12:35:51.865] Disconnected
[12:35:52.377] Connected to /dev/rfcomm0
[12:35:52.596] Disconnected

```

Zackariyya, 2024-12-22:
Hey, Booting my arch linux install and installing python-nxt via ``yay nxt-python'' and `` pacman -S bluez bluez-utls bluez-deprecated-tools``, the program works as expected, so maybye my Fedora install is broken ...

schodet, 2024-12-23:
Locking at the strace output, the connection failed with EACCES (Permission denied)… so there is something forbidding the connection. You can try with the btsocket branch (see #199), but I think the result will be the same.

Thank you very much for your tests results. If you can find what is blocking on the Fedora install, this would make an interesting addition to the manual.

Maybe there is something interesting in the machine log files.

schodet, 2024-12-23:
Actually, on https://github.com/pybluez/pybluez/issues/473, the reporter said that it works using direct sockets, so testing with the btsocket branch could be interesting too.

Thanks.

schodet, 2025-01-10:
@Zackariyya could you look at Fedora log files to see if there is some useful information there?

Did you had the opportunity to try the btsocket branch?

Zackariyya, 2025-06-12:
Hi,
I actually erased my computer and installed arch, so i cant debug any further, sorry!