schodet/nxt-python #201: Status of bluetooth connection on later macos version

erard22, 2025-06-29:
- OS: MacOs 15.5 (24F74)
- Python version: 3
- NXT-Python version: latest
- Connection method: USB? Bluetooth? Bluetooth

Hello, 

I just wanted to try out the library to use an old nxt via bluetooth with my son.
The nxt has a broken display, so python and bluetooth could be the alternative. As there is no windows computer in the house.
But what's the status here? I followed the steps on the installation page, nxt is connected via bluetooth, but no brick is found. Via usb cable the script is working.

nxt.locator.BrickNotFoundError: no brick found

Anyone else is able to use bluetooth?

RZI3D, 2025-06-29:
Did you install [PyBlueZ?](https://pybluez.readthedocs.io/en/latest/install.html)

schodet, 2025-07-02:
Problem is that pybluez is no longer supported, and this is complicated to install it on macos.

See here: https://ni.srht.site/nxt-python/latest/installation.html#installation-using-pip

Another option would be to use a bluetooth socket, but nobody tried it on MacOS yet: https://github.com/schodet/nxt-python/issues/199#issuecomment-2558238693

If you want to try, please test the btsocket branch and report the result.

Also documentation could be improved by providing a test-matrix of which OS works with which setup, this could also be used by nxt-python to select the best option according to the OS.

schodet, 2025-07-03:
Thanks @dlech (and sorry if you did mention it earlier but I forgot).

So only option today for MacOS is to manage to build pybluez from source.